sonic-net/sonic-linux-kernel #117: make error:cannot run debian/linux-cpupower.install: No such file or directory

dingrui37, 2019-12-17:
ubuntu 16.04
```
Can't exec "/home/dingrui/projects/sonic-linux-kernel/linux-4.9.168/debian/linux-cpupower.install": No such file or directory at /usr/share/perl5/Debian/Debhelper/Dh_Lib.pm line 776.
dh_install: cannot run debian/linux-cpupower.install: No such file or directory
make[3]: *** [install-source] Error 2
make[2]: *** [binary-indep] Error 2
make[1]: *** [binary-indep] Error 2
make: *** [/linux-headers-4.9.0-9-2-common_4.9.168-1+deb9u5_all.deb] Error 2
```

timmy00274672, 2020-05-25:
I hit the same issue as well

antony-rheneus, 2020-06-11:
dh-exec packagec is removed during sonic-slave-stretch build along with other packages.
 I suspect it because of the stretch-backports.

Need to override the apt auto remove behaviour

Removing gem2deb (0.33.1) ...
Removing kernel-wedge (2.96) ...
Removing dh-exec (0.23+b1) ...
Removing dh-make (2.201608) ...
Removing dh-systemd (10.2.5) ...
Removing equivs (2.0.9+nmu1) ...
Removing javahelper (0.59) ...
Removing pkg-php-tools (1.35) ...
Removing python-stdeb (0.8.5-1) ...
Removing dh-strip-nondeterminism (0.034-1) ...
Removing debhelper (12.1.1~bpo9+1) ...
Removing dh-autoreconf (17~bpo9+1) ...

yzwddsg-plus, 2025-03-21:
I hit this too
Can't exec "/tmp/linux-5.10.197/debian/linux-cpupower.install": No such file or directory at /usr/share/perl5/Debian/Debhelper/Dh_Lib.pm line 1538.
dh_install: error: cannot run debian/linux-cpupower.install: No such file or directory

yzwddsg-plus, 2025-03-21:
> I hit this too Can't exec "/tmp/linux-5.10.197/debian/linux-cpupower.install": No such file or directory at /usr/share/perl5/Debian/Debhelper/Dh_Lib.pm line 1538. dh_install: error: cannot run debian/linux-cpupower.install: No such file or directory

I have successfully resolved the issue by executing the command chmod -x debian/*.install
from: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=901715