sonic-net/sonic-linux-kernel #144: ABI check in Buster

olivier-singla, 2020-06-04:
In SONiC branch 201911 (Stretch), there is this patch which is applied:

packaging-update-abiname-to-11-2.patch

It changes the kernel ABI from 11 to 11-2

This has the side effect to disable the ABI check performed by the script
debian/bin/buildcheck.py

This is because the buildcheck.py script is looking to compare the file
debian/build/build_amd64_none_amd64/Module.symvers
(generated by the build)
with the file
debian/abi/4.9.0-11-2/amd64_none_amd64

But since the ABI has changed, the Debian file checked-in is still:
debian/abi/4.9.0-11/amd64_none_amd64

So buildcheck.py, not finding the file
debian/abi/4.9.0-11-2/amd64_none_amd64
just bails out, without error, and will not check the ABI change.

In SONiC latest master branch, based on Buster, there is no such patch to change the ABI.
The script buildcheck.py will then compare the file
debian/build/build_amd64_none_amd64/Module.symvers
with the file
debian/abi/4.19.0-6/amd64_none_amd64

so any ABI change will be effectively checked..

For example, the patch:
Support-for-fullcone-nat.patch
which apply fine with SONiC 201911 (Stretch) can not currently be applied in the master branch (Buster), because there is a change in a global kernel structure
(include/net/netfilter/nf_conntrack.h in this example).

I am wondering, what is the plan regarding the ABI in SONiC Buster, are we going to change it like this was done in SONiC Strech?

Without changing the kernel ABI, many patches might just fail because the ABI check...

lguohan, 2020-06-04:
yes, if there is need to change ABI, we can follow the same practice as stretch kernel approach.

olivier-singla, 2020-06-05:
I created these two PRs to address the kernel ABI change:
- https://github.com/Azure/sonic-buildimage/pull/4711
- https://github.com/Azure/sonic-linux-kernel/pull/145

closed 2020-06-26 by lguohan