sonic-net/sonic-linux-kernel #175: Kernel download links are invalid

Str1ker17, 2020-11-26:
See https://github.com/Azure/sonic-linux-kernel/blob/8eee9d8f7ba51fde40b8ae0dd3fab99c935a3d98/Makefile#L28 for example

`LINUX_IMAGE_URL` resolves to 
`https://sonicstorage.blob.core.windows.net/packages/kernel-public/linux-image-4.19.0-9-2-amd64_4.19.118-2+deb10u1_amd64.deb?sv=2015-04-05&sr=b&sig=oRGGO9xJ6jmF31KGy%2BwoqEYMuTfCDcfILKIJbbaRFkU%3D&se=2128-12-23T19%3A06%3A47Z&sp=r`
in the build time.

I get `403 Server failed to authenticate the request. Make sure the value of Autho
rization header is formed correctly including the signature.` error message from wget.

If I try to download the file from the browser, the result is the same 403.

Please make these links work as rebuilding the kernel from sources is a very time-consuming process and I'd like to just download it.

saiarcot895, 2024-07-02:
Fixed in #400. Pre-compiled kernel downloads are no longer available, and so have been removed.

closed 2024-07-02 by saiarcot895

Str1ker17, 2024-07-02:
Why it was deleted? Compiling the kernel locally every time is really slow...