sonic-net/sonic-linux-kernel #343: Problems encountered by optoe drivers

philo-micas, 2023-10-11:
Code Path：drivers/misc/eeprom/optoe.c
Issue 1：The write_timeout value is fixed at 25ms. In the test of our products, it is found that the 25ms timeout time of some optical modules is not enough, and I2C read and write failures will occur occasionally.
Issue 2：When switch optoe’s dev_class from 2 to 1 then 1 to 3, clash occurs.

Regarding this issue, our solution is:

saiarcot895, 2023-10-17:
@prgeor @donboll would one of you be able to take a look at this? I know issue 2 was reported in #317 as well.

prgeor, 2024-05-02:
@philo-micas please check this fix https://github.com/sonic-net/sonic-linux-kernel/blob/master/patch/driver-support-optoe-dynamic-write-timeout.patch

closed 2024-07-16 by saiarcot895